ImSwitch/ImSwitch #193: Cool Led Command

niallmacq, 2023-11-22:
I have a setup with a tis camera, a coolLed pE-300ultra and a marhauzer stage. I have tried to create a json file and am attempting to modify the existing codebase to allow coolLed control by modifying the command to the format expected. The commands are here: https://www.coolled.com/wp-content/uploads/2018/09/CoolLED-Commands-Manual-DOC-038-Iss-2.pdf 
I modified it like this:

def setEnabled(self, enabled):
        """Turn on (N) or off (F) laser emission"""
        if enabled:
            value = "N"
        else:
            value = "F"
        cmd = "CSS" + self.__channel_index + value
        self._rs232manager.query(cmd)

Is this correct. Also, It says I have a mock RS232 initialised. Does this mean the commands aren't being sent?
here is my setup json file. Am I doing something wrong?
[coolLED.json](https://github.com/ImSwitch/ImSwitch/files/13437879/coolLED.json)

jacopoabramo, 2023-11-22:
Hi @niallmacq, thanks for reaching out. From what I see it seems that it could be a problem related to the COM port to which the CoolLED is connected. Could you post the log output that it's shown in the command line where you're launching ImSwitch after startup?

niallmacq, 2023-11-22:
Hi @jacopoabramo, 

Thanks for your quick reply.

I've added that line and the following is output.
2023-11-22 10:57:43 DEBUG [ImConMainController] Initializing
2023-11-22 10:57:43 WARNING [RS232Manager -> coolLED] Initializing mock RS232 port
2023-11-22 10:57:44 DEBUG [CoolLEDLaserManager -> V LED] CSSASN000

beniroquai, 2023-11-22:
Indeed, the Coooled driver cannot initialize the device. Can you check the COM port in the windows device manager? Also, can you connect to the device using the coolled software?

niallmacq, 2023-11-22:
I have used micromanager and it connects on com6. On device manager it show up on com6 and com7.

jacopoabramo, 2023-11-22:
Could you try setting COM7 on the configuration file then? Or is it connected to something else?

niallmacq, 2023-11-22:
No, still the same. I looked in the coolLed manual and they say there are 2 ports to allow control and monitoring of the device in parallel. You can use either ports for control.

The TisManager isn't working either. It's not detecting any cameras. Could it be something about how I've installed ImSwitch?

Could it be something about my installation?

jacopoabramo, 2023-11-22:
I don't think it's an installation issue rather than a configuration file problem.

Camera wise:

- are you able to use your TIS camera outside of ImSwitch with its own capture software?
- could you also copy here the entire log that pops out from the command line, including everything related to the camera as well?

CoolLED wise:

- if you revert the control from "CSS" to "C" does it work normally?

jacopoabramo, 2023-11-22:
It doesn't show any warning related to the TISManager though. Are you using the correct configuration?

jacopoabramo, 2023-11-22:
For the CoolLED: without access to the hardware I cannot verify what's going on with the communication between the RS232 manager and the device. Is there any chance you could setup a debug environment and verify what happens during initialization? If you're unfamiliar with this we can setup a remote call on a private channel and try to figure it out together.

jacopoabramo, 2023-11-22:
> I think imswitch is not initialising the serial port correctly

You could try going in the `RS232Manager` class, in the [`_getRS232port`](https://github.com/ImSwitch/ImSwitch/blob/cc7cb102ae62858ce85c1bdc3baaa3e6c46469bd/imswitch/imcontrol/model/managers/rs232/RS232Manager.py#L45) and add some more logging to see what kind of exception is thrown. Something like:

```py
    def _getRS232port(self, port, settings):
        try:
            from imswitch.imcontrol.model.interfaces.RS232Driver import generateDriverClass
            DriverClass = generateDriverClass(settings)
            rs232port = DriverClass(port)
            rs232port.initialize()
            return rs232port
        except Exception as e:
            self.__logger.exception(f"Thrown exception {e}")
            self.__logger.warning('Initializing mock RS232 port')

            from imswitch.imcontrol.model.interfaces.RS232Driver_mock import MockRS232Driver
            return MockRS232Driver(port, settings)
```

niallmacq, 2023-11-22:
2023-11-22 16:41:59 DEBUG [ImConMainController] Initializing
2023-11-22 16:41:59 ERROR [RS232Manager -> coolLED] Thrown exception <class 'Exception'>
Traceback (most recent call last):
  File "C:\Users\Cairn Research\AppData\Roaming\Python\Python39\site-packages\imswitch\imcontrol\model\managers\rs232\RS232Manager.py", line 49, in _getRS232port
    rs232port = DriverClass(port)
  File "C:\Users\Cairn Research\AppData\Roaming\Python\Python39\site-packages\imswitch\imcontrol\model\interfaces\RS232Driver.py", line 9, in __init__
    super().__init__(port)
  File "C:\Users\Cairn Research\AppData\Roaming\Python\Python39\site-packages\lantz\core\messagebased.py", line 377, in __init__
    self.__resource_manager = get_resource_manager()
  File "C:\Users\Cairn Research\AppData\Roaming\Python\Python39\site-packages\lantz\core\messagebased.py", line 37, in get_resource_manager
    _resource_manager = visa.ResourceManager(config.VISA_BACKEND)
AttributeError: 'str' object has no attribute 'ResourceManager'
2023-11-22 16:41:59 WARNING [RS232Manager -> coolLED] Initializing mock RS232 port

jacopoabramo, 2023-11-23:
So the log you copied shows that there's something going wrong with this line in the lantz core:

```
File "C:\Users\Cairn Research\AppData\Roaming\Python\Python39\site-packages\lantz\core\messagebased.py", line 37, in get_resource_manager
_resource_manager = visa.ResourceManager(config.VISA_BACKEND)
```

For some reason the `ResourceManager` is receving a string object and not the correct `config.VISA_BACKEND` object. This could be caused by the fact that the `MessageBasedDriver` class imported in [this line](https://github.com/ImSwitch/ImSwitch/blob/cc7cb102ae62858ce85c1bdc3baaa3e6c46469bd/imswitch/imcontrol/model/interfaces/RS232Driver.py#L1) is not doing so correctly. You could try changing it with

```py
from lantz.core.messagebased import MessageBasedDriver
```

This is a more straightforward import and maybe fixes the issue, but I'm not 100% sure and would still need to debug this at run time.

jacopoabramo, 2023-11-27:
After trying to hack the lantz.core.messagebased file in order to substitute the `visa` import with `pyvisa` this is what we obtained:

```
Traceback (most recent call last):
  File "C:\Users\Niall\anaconda3\envs\imswitchdev\lib\site-packages\pyvisa_py\highlevel.py", line 156, in open
    parsed = rname.parse_resource_name(resource_name)
  File "C:\Users\Niall\anaconda3\envs\imswitchdev\lib\site-packages\pyvisa\rname.py", line 231, in from_string
    raise InvalidResourceName(
pyvisa.rname.InvalidResourceName: Could not parse COM6: unknown interface type

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Niall\Documents\ImSwitchclone\imswitch\imcontrol\model\managers\rs232\RS232Manager.py", line 50, in _getRS232port
    rs232port.initialize()
  File "C:\Users\Niall\Documents\ImSwitchclone\imswitch\imcontrol\model\interfaces\RS232Driver.py", line 31, in initialize
    super().initialize()
  File "C:\Users\Niall\AppData\Roaming\Python\Python39\site-packages\lantz\core\messagebased.py", line 413, in initialize
    self.resource = get_resource_manager().open_resource(self.resource_name, **self.resource_kwargs)
  File "C:\Users\Niall\anaconda3\envs\imswitchdev\lib\site-packages\pyvisa\highlevel.py", line 3292, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\Niall\anaconda3\envs\imswitchdev\lib\site-packages\pyvisa\resources\resource.py", line 281, in open
    self.session, status = self._resource_manager.open_bare_resource(
  File "C:\Users\Niall\anaconda3\envs\imswitchdev\lib\site-packages\pyvisa\highlevel.py", line 3217, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Users\Niall\anaconda3\envs\imswitchdev\lib\site-packages\pyvisa_py\highlevel.py", line 160, in open
    self.handle_return_value(None, StatusCode.error_invalid_resource_name),
  File "C:\Users\Niall\anaconda3\envs\imswitchdev\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_INV_RSRC_NAME (-1073807342): Invalid resource reference specified. Parsing error.
```

pretty sure this doesn't help much; at any rate the lantz package seems to be quite old, but I'll open an issue there as well

jacopoabramo, 2023-11-28:
Hi @niallmacq , happy it works. Did you hack further the RS232Manager or did you just change the JSON file and now everything works?

kasasxav, 2023-11-28:
Yes exactly, they changed the library and now instead of COMXX you have to write it that way in the json

jacopoabramo, 2023-11-28:
Ah, good to know. We'll have to change the pre-shipped JSONs then. Thanks for the help @kasasxav , closing.

closed 2023-11-28 by jacopoabramo