ImSwitch/ImSwitch #195: Creating and sharing new attributes to enrich metadata

jacopoabramo, 2023-12-13:
I'm currently in contact with other parties that are working on ImSwitch to implement a new set of Widget/Controllers for their application. One thing I'm trying to wrap my mind around is to add new shared attributes on a general controller in order to give the possibility to developers to add their own set of metadata. @kasasxav it is not very clear to me from the documentation how to achieve this. Could you give me a brief rundown? I'll add the notes to the documentation in a more formatted way.

kasasxav, 2023-12-13:
Hi @jacopoabramo , check the class [sharedAttributes](https://github.com/ImSwitch/ImSwitch/blob/b05a431320671c2af16b6db35bed6818d4826d70/imswitch/imcommon/model/SharedAttributes.py#L6), and how it's implemented in the different classes.

All widget controllers by default have the function [setSharedAttr](https://github.com/ImSwitch/ImSwitch/blob/b05a431320671c2af16b6db35bed6818d4826d70/imswitch/imcontrol/controller/basecontrollers.py#L223), so if you do that with new data it will automatically be included as metadata in the hdf5 file.

You can see how it's done in any of the controllers, for example the [LaserController](https://github.com/ImSwitch/ImSwitch/blob/b05a431320671c2af16b6db35bed6818d4826d70/imswitch/imcontrol/controller/controllers/LaserController.py#L239).

It's probably not included in the Documentation, feel free ot include it :)

kasasxav, 2023-12-13:
Actually better check in [PositionersController](https://github.com/ImSwitch/ImSwitch/blob/b05a431320671c2af16b6db35bed6818d4826d70/imswitch/imcontrol/controller/controllers/PositionerController.py), the Laser one is weird.

`for axis in pManager.axes:
                self.setSharedAttr(pName, axis, _positionAttr, pManager.position[axis])
                if speed:
                    self.setSharedAttr(pName, axis, _positionAttr, pManager.speed)`

jacopoabramo, 2023-12-13:
Awesome, thanks. I'll try to document this and then close the issue.

beniroquai, 2024-01-13:
Tested. Works! Thanks a lot @kasasxav! :) 
@jacopoabramo would it make sense to refactor the structure to make this compatible with existing "standards" á la FAIR?