ImSwitch/ImSwitch #197: BeadRec Widget not displaying properly

nicktoledog, 2024-01-11:
**Bug description**
When running the BeadRec widget the computed image is not displayed, just showing pure black. The terminal shows the following error:

```
 ERROR [ExceptionHandler] Traceback (most recent call last):
  File "C:\Users\azken\anaconda3\envs\scream-imswitch\Lib\site-packages\pyqtgraph\graphicsItems\ImageItem.py", line 866, in paint
    self.render()
  File "C:\Users\azken\anaconda3\envs\scream-imswitch\Lib\site-packages\pyqtgraph\graphicsItems\ImageItem.py", line 583, in render
    qimage = self._try_make_qimage(image, levels, lut, augmented_alpha)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\azken\anaconda3\envs\scream-imswitch\Lib\site-packages\pyqtgraph\graphicsItems\ImageItem.py", line 855, in _try_make_qimage
    raise ValueError("unsupported image type")
ValueError: unsupported image type
```

**How to reproduce**
Steps to reproduce the behavior:
1. Select the BeadRec widget.
2. Click the Select ROI button of the widget.
3. Click the Run box of the widget.

**Expected behavior**
The image displayed at the widget should be updating continously while the Run box is selected.

**Environment**
 - OS: Windows 10
 - ImSwitch version: 2.0.0

**Additional context**
This widget worked well in the previous Imswitch version.

beniroquai, 2024-01-12:
Hey @nicktoledog, thanks for reporting the error. I must admit that I have never used this widget. Just activated it and get this gui element:

[image]

Do you get that far? When activating it, I'll get the following next error - is this related/does it show up on your end as well? I will start a scan instance and try to reproduce your error then.

jonatanalvelid, 2024-01-15:
Hi, Unfortunately I have not written anything on nor used the BeadRec widget. @kasasxav would be the one that knows better about that.

kasasxav, 2024-01-15:
Hi @nicktoledog, 

The beadrec widget is connected to the scan widget. The idea is that you do a bead scan and the beadrec for each scan position will sum all the values of the scan image and allocate it in the final image, which will have the same size as the scan parameters.

Are you also using the scan widget? If so, could you send a screenshot of it to see which parameters you use?
If not, how are you scanning the beads? The widget could be modified to not depend on the scan widget, I think it's mostly the image width and height that is used.

nicktoledog, 2024-01-16:
Hi! Thanks a lot for your responses!

I indeed get to the screen that @beniroquai attaches. Regarding the scan, @kasasxav, I do not use the scan widget to perform the scan, I use a LabVIEW script that sends the trigger to acquire an image every scan step. I tried filling the scan widget with my parameters however the computed image is still not displayed.

[image]

And I still get the same error.

```
 ERROR [ExceptionHandler] Traceback (most recent call last):
  File "C:\Users\azken\anaconda3\envs\scream-imswitch\Lib\site-packages\pyqtgraph\graphicsItems\ImageItem.py", line 866, in paint
    self.render()
  File "C:\Users\azken\anaconda3\envs\scream-imswitch\Lib\site-packages\pyqtgraph\graphicsItems\ImageItem.py", line 583, in render
    qimage = self._try_make_qimage(image, levels, lut, augmented_alpha)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\azken\anaconda3\envs\scream-imswitch\Lib\site-packages\pyqtgraph\graphicsItems\ImageItem.py", line 855, in _try_make_qimage
    raise ValueError("unsupported image type")
ValueError: unsupported image type
```

kasasxav, 2024-01-16:
Hi,

Alright, thank you. Are you running ImSwitch with the source version or pip install?
Could you add some lines of code and let me know about the output?
Also, another question, which Camera manager are you using? Could you see if it has the function: getChunk implemented?

in imswitch/imcontrol/controller/controllers/BeadRecController.py
In line 7 (in the imports):

`from imswitch.imcommon.model.logging import initLogger`

In line 73 (init function of BeadWorker)
`self.__logger = initLogger(self, tryInheritParent=False)`

after line 82, could you: 
```
self.__logger.info(type(newImages))
self.__logger.info(newImages)

```
My guess is that getChunk() is returning None because if you're not using the camera with the scan maybe you don't need it (that's a function we have to save images into files); then the image type is wrong because it's None. When you say it was working in other versions, I'm hoping maybe you were using also another camera manager?  That's my hypothesis :)

kasasxav, 2024-01-16:
Since you are adding that could you add a couple more to check if the issue comes from the scanning:

in Line 23 (init of BeadRecController)
`self.__logger = initLogger(self, tryInheritParent=False)`

in line 54 (run function of BeadRecController)
`self.__logger.info(self.dims)`

nicktoledog, 2024-01-17:
Hi! I am running the version obtained with pip install. Regarding the Camera manager I am using the HamamatsuManager and indeed has the following  getChunk function:

```
def getChunk(self):
        return self._camera.getFrames()[0]
```
I added the lines you mentioned and I get the following.

```
2024-01-17 12:14:09 INFO [BeadRecController] [10 10  5]
2024-01-17 12:14:09 INFO [BeadWorker] <class 'numpy.ndarray'>
2024-01-17 12:14:09 INFO [BeadWorker] [[[111 104  98 ..., 105 117  99]
  [102 102  92 ..., 106  95 102]
  [ 95 109 102 ..., 105  97 100]
```

Also noticed that whe I was using the full chip and not selecting a ROI in the Detector settings this was the output:

```
2024-01-17 12:11:04 INFO [BeadRecController] [10 10  5]
2024-01-17 12:11:04 INFO [BeadWorker] <class 'numpy.ndarray'>
2024-01-17 12:11:04 INFO [BeadWorker] []
```
It eventually outputted an array as the one above.

However the same error keeps happening and no image is visible within the widget, I have the same scan parameters as the ones in the already uploaded screenshot.

```
2024-01-17 12:18:40 ERROR [ExceptionHandler] Traceback (most recent call last):
  File "C:\Users\azken\anaconda3\envs\scream-imswitch\Lib\site-packages\pyqtgraph\graphicsItems\ImageItem.py", line 866, in paint
    self.render()
  File "C:\Users\azken\anaconda3\envs\scream-imswitch\Lib\site-packages\pyqtgraph\graphicsItems\ImageItem.py", line 583, in render
    qimage = self._try_make_qimage(image, levels, lut, augmented_alpha)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\azken\anaconda3\envs\scream-imswitch\Lib\site-packages\pyqtgraph\graphicsItems\ImageItem.py", line 855, in _try_make_qimage
    raise ValueError("unsupported image type")
ValueError: unsupported image type
```
I also noticed that when I run the scan widget this message appears, I do not now if it is of any help for this issue but here it is just in case.

```
2024-01-17 12:21:16 ERROR [ScanController] Traceback (most recent call last):
  File "C:\Users\azken\anaconda3\envs\scream-imswitch\Lib\site-packages\imswitch\imcontrol\controller\controllers\ScanController.py", line 205, in runScanAdvanced
    self.signalDict, self.scanInfoDict = self._master.scanManager.makeFullScan(
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\azken\anaconda3\envs\scream-imswitch\Lib\site-packages\imswitch\imcontrol\model\managers\ScanManager.py", line 96, in makeFullScan
    scanSignalsDict, positions, scanInfoDict = self.getScanSignalsDict(scanParameters)
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\azken\anaconda3\envs\scream-imswitch\Lib\site-packages\imswitch\imcontrol\model\managers\ScanManager.py", line 82, in getScanSignalsDict
    return self._scanDesigner.make_signal(parameterDict, self._setupInfo)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\azken\anaconda3\envs\scream-imswitch\Lib\site-packages\imswitch\imcontrol\model\signaldesigners\BetaScanDesigner.py", line 90, in make_signal
    rampSignal[end - smooth - settling: end - settling] = self.__smoothRamp(rampValues[s], rampValues[s + 1], smooth)
    ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: could not broadcast input array from shape (100,) into shape (4,)

```

Thanks a lot!

nicktoledog, 2024-01-18:
I added the following lines since it gave me a concatenation error:
```
im = np.resize(self.recIm, self.dims + 1)
self.__logger.info("dims: " + str(self.dims))
self.__logger.info("type(recIm): " + str(type(self.recIm)))
self.__logger.info("type(im): "+ str(type(im)))
self.__logger.info("im: " + str(im))
self._widget.updateImage(im)
```
And the output new output is the follwing:
```
2024-01-18 15:18:36 INFO [BeadRecController] dims: [10 10  5]
2024-01-18 15:18:36 INFO [BeadRecController] type(recIm): <class 'numpy.ndarray'>
2024-01-18 15:18:36 INFO [BeadRecController] type(im): <class 'numpy.ndarray'>
2024-01-18 15:18:36 INFO [BeadRecController] im: [[[ 177.0769043   177.45776367  177.01806641  176.58398438  177.3527832
    177.12304688]
  [ 176.59472656  176.96240234  177.26220703  176.85375977  176.6171875
    176.60571289]
  [ 176.83666992  177.65307617  177.00952148  177.35253906  176.57836914
    177.20849609]
  [ 176.86474609  177.15209961  176.56054688  177.19580078  177.09570312
    176.93652344]
  [ 177.90673828  177.43359375  177.67895508  177.79882812  177.66967773
    177.05151367]
  [ 177.51928711  177.33178711  177.89916992  177.53442383  177.37646484
    176.86010742]
```
Still the same "ValueError: unsupported image type" keeps appearing :(

kasasxav, 2024-01-18:
Alright, thank you @nicktoledog and sorry for making you do all of this debugging (I cannot test these things anymore).

It seems that everything is right to me, from the BeadController side... I'm thinking it could maybe be that the image is 3D and pyqtgraph only takes 2D images? Could you try with a 2D scan? And changing also the parameters in the widget?

Otherwise it's definitely something in the library, in line 61 here: https://github.com/ImSwitch/ImSwitch/blob/master/imswitch/imcontrol/view/widgets/BeadRecWidget.py
in that case, could you debug it further @jacopoabramo ? Maybe let's try to see first if it works with 2D but otherwise it's something external.

nicktoledog, 2024-01-19:
No problem @kasasxav! Thanks a lot for helping me! In fact what I was trying to do was a 2D scan, but it seems that even though I put the Z parameter to None for some reason the output dimensions are still 3D as in the previous posts. The widget works properly when the Z size is set to 0 and the dimension is 3rd dimension is kept!

[image]

These are the new outputs:

```
2024-01-19 11:09:05 INFO [BeadRecController] [20 20]
2024-01-19 11:09:05 INFO [BeadWorker] <class 'numpy.ndarray'>
2024-01-19 11:09:05 INFO [BeadWorker] []

2024-01-19 11:09:05 INFO [BeadRecController] dims: [20 20]
2024-01-19 11:09:05 INFO [BeadWorker] <class 'numpy.ndarray'>
2024-01-19 11:09:05 INFO [BeadRecController] type(recIm): <class 'numpy.ndarray'>
2024-01-19 11:09:05 INFO [BeadWorker] []
2024-01-19 11:09:05 INFO [BeadRecController] type(im): <class 'numpy.ndarray'>
2024-01-19 11:09:05 INFO [BeadRecController] im: [[ 104.20760234    0.            0.            0.            0.
     0.            0.            0.            0.            0.
     0.            0.            0.            0.            0.
     0.            0.            0.            0.            0.
     0.        ]
 [   0.            0.            0.            0.            0.
     0.            0.            0.            0.            0.
     0.            0.            0.            0.            0.
     0.            0.            0.            0.            0.
```
Thanks a lot for all your help!! :)

kasasxav, 2024-01-19:
Alright @nicktoledog , so we finally figured it out !! Happy to help, sorry it took some time, in the beginning I really thought it would be an issue from the dimensions or the image being None.

Definitely this widget has lots of potential improvements ahead:
* Make it not necessarily dependant on the scan widget. For example one could include the dimensions of the scan in the same widget or have a checkbox "Get dimensions from scan".
* Include support for 3D scans.

I made this widget more for myself and since not so many people used it I didn't really make it general enough.

If you feel like contributing to ImSwitch and doing some merge request for it, it would be amazing. Otherwise maybe someone else can work on it when they have time.

Can I close the issue for now?

closed 2024-02-15 by kasasxav