ImSwitch/ImSwitch #208: The Imaging Source IC4 SDK integration

Kubiczek36, 2024-06-21:
The Imaging source has published a new SDK, which offers more features, like acsess to the buffer or setting the triggers.  

## Usefull links 
- [IC4 Docs](https://www.theimagingsource.com/en-us/documentation/ic4python/index.html)
- [IC4 examples repo](https://github.com/TheImagingSource/ic4-examples)

Kubiczek36, 2024-06-21:
## Current issues

* after switching the LiveView on and off the ImSwitch crashes
* after opening and closing the window with properties the ImSwitch crashes

jacopoabramo, 2024-06-24:
I worked with another developer on a fork that gives a new implementation of the Imaging Source cameras based off the IC4 SDK. I still need to create a PR on the dev branch and push it. I think we managed to have a working live view implementation.

Relevant code as follows:

- a manager implementation [here](https://github.com/palec87/ImSwitchOpt/blob/add-camera-mock/imswitch/imcontrol/model/managers/detectors/TIS4Manager.py)

- interface details [here](https://github.com/palec87/ImSwitchOpt/blob/add-camera-mock/imswitch/imcontrol/model/interfaces/tis4camera.py)

Kubiczek36, 2024-06-25:
@jacopoabramo I've checked, what you did and I would be happy to join forces. There is only one but, and that is the Sink you use. In my implementation I use a Queue sink, where you define a acquisition params (also trigger for example) and what happends with every frame that the camera acquires. I store the frames in a list from which the recording/liveview worker can pick it up. On the other hand you use SnapSink where you must call a method to acquire the image.

jacopoabramo, 2024-06-26:
@Kubiczek36 I'm currently reviewing the PR involved in including the IC4 SDK. I have to admit that I'm not familiar with it because I do not have access to the hardware to test it out. 

@palec87 you implemented the IC4 interface in your PR, how hard would it be to integrate the changes requested in this issue?

palec87, 2024-06-26:
Hi @Kubiczek36 and @jacopoabramo. This is great and necessary addition. Currently we don't do hardware trigger, so I cannot test that, however QueueSink I can check. The question is how to merge the two. 

* I think overall, the @Kubiczek36 IC4 is extending the OPT snapping. If the Queue sink is done, let' try to merge and test it after the OPT PR is done. Then both sink and queue could work.

Kubiczek36, 2024-07-02:
Can I just ask you, what do you mean with OPT snapping and OPT PR?

palec87, 2024-07-02:
> Can I just ask you, what do you mean with OPT snapping and OPT PR?

Hi, sorry @Kubiczek36, that is the PR which Jacopo mentioned is almost finished to be merged to the main imswitch branch. It has the snapping functionality of the ic4 implemented, however without the QueueSink, but hopefully will not be difficult to merge your work to it.