ImSwitch/ImSwitch #217: Double Log-output - why?

beniroquai, 2024-08-21:
@jacopoabramo for some reasons all logs come duplicated. One in colour, one without. I remember that at one point I touched the `tryInheritParent` setting, but I cannot find differences between old ImSwitch master code and mine. Any ideas? (CC @kasasxav 😉)

[image]

jacopoabramo, 2024-08-21:
Yes I notice this pretty often as well. Just out of curiosity how do you launch ImSwitch in this scenario?

beniroquai, 2024-08-22:
I can launch it in any way e.g. 
`python3 -m imswitch`, 
`imswitch` 
.. .
The result is the same. Maybe some settings that went off with the coloured logs?