ImSwitch/ImSwitch #218: Fail to start example_sted configuration

sbancelin, 2024-08-22:
Hello,
I am trying to test the example_sted.json configuration. Unfortunatly, when I start Imswitch I got an error message coming from the detector settings. It seems the params.binning.value is giving an empty string. 
Am I the only encountering this issue?

  File "C:\ImSwitch\imswitch\__main__.py", line 51, in main
    view, controller = modulePkg.getMainViewAndController(
  File "C:\ImSwitch\imswitch\imcontrol\__init__.py", line 58, in getMainViewAndController
    raise e
  File "C:\ImSwitch\imswitch\imcontrol\__init__.py", line 54, in getMainViewAndController
    controller = ImConMainController(options, setupInfo, view, moduleCommChannel)
  File "C:\ImSwitch\imswitch\imcontrol\controller\ImConMainController.py", line 54, in __init__
    self.controllers[widgetKey] = self.__factory.createController(
  File "C:\ImSwitch\imswitch\imcommon\controller\basecontrollers.py", line 46, in createController
    controller = controllerClass(*self.__args, *args,
  File "C:\ImSwitch\imswitch\imcontrol\controller\controllers\SettingsController.py", line 57, in __init__
    execOnAll(lambda c: (self.adjustFrame(detector=c)),
  File "C:\ImSwitch\imswitch\imcontrol\model\managers\MultiManager.py", line 55, in execOnAll
    return {managedDeviceName: func(subManager)
  File "C:\ImSwitch\imswitch\imcontrol\model\managers\MultiManager.py", line 55, in <dictcomp>
    return {managedDeviceName: func(subManager)
  File "C:\ImSwitch\imswitch\imcontrol\controller\controllers\SettingsController.py", line 57, in <lambda>
    execOnAll(lambda c: (self.adjustFrame(detector=c)),
  File "C:\ImSwitch\imswitch\imcontrol\controller\controllers\SettingsController.py", line 161, in adjustFrame
    binning = int(params.binning.value())
ValueError: invalid literal for int() with base 10: ''

**Environment**
 - OS: Windows 11
 - ImSwitch version: 2.0.0, master branch
 - How you downloaded ImSwitch: using pip install -r requirements-dev.txt

jacopoabramo, 2024-08-22:
Greetings, yes I'm aware of this. It's a bug due to the latest changes of the pyqtgraph package and its usage in the `SettingsWidget` and `SettingsController` class.

It should be fixable by installing a previous version of the package. I believe it should work until version 0.13.3

Try doing

`pip uninstall pyqtgraph`

then 

`pip install pyqtgraph<0.13.4`

Let me know if this fixes the problem.

sbancelin, 2024-08-22:
Great, problem solved! As you said, it works with pyqtgraph 0.13.3.

Many thanks for your quick reply.

closed 2024-08-22 by sbancelin