ImSwitch/ImSwitch #223: Looking for new project ownership

jacopoabramo, 2025-01-08:
Hi everyone, and happy new year.

I'm opening this issue to make an announcement. I will not be following the ImSwitch project anymore and I'm actively looking to transfer ownership of the project to other interested parties.

The reasons for my decision are purely personal and I will not disclose them in public. I believe anyway that the time for me to work on this project is over, and I would like to work on something different at this time. I apologize for any inconvenience this decision might cause.

I will transfer ownership only to developers who I know contributed in the past to the project.

If you're interested in taking ownership of the project please reach out to me over this issue.

xcasas, 2025-01-15:
Hi @jacopoabramo , I understand. From my side, I also want to focus on my current projects and therefore continue with my decision of leaving ImSwitch behind.

@GuillaumeMinet @andreavolpato @fbalzarotti , etc - I tag you in case you are interested to organize the mantainance of ImSwitch, since you will probably use it to some degree.

beniroquai, 2025-01-15:
Hey! I I discuss that with Jacopo already and I think I'm rather waiting for the next bigger thing that Jacopo is proposing ;-) 
I am still heavily working on Ipswich and trying to make it more mobile friendly. But since the branch of the UC two is so far off the track, I don't think it makes a lot of sense to try to keep the main repository up-to-date since it will be very specific for server client infrastructures. Since there is not so much traction here anymore anyway I don't think there's a lot of work to be done from any person who is taking over. But this is just my two cents..
Which is and was a great piece of software.
And I am keeping on using it. Whoever is keen on using it in the web, please have a look here: https://imswitch.openuc2.com/imswitch/index.html

LREIN663, 2026-10-13:
Hi @jacopoabramo, thanks for all the work you’ve put into ImSwitch!
Following our recent discussion, I’d like to express our interest here publicly: my two colleagues, @GuillaumeMinet  and @khoj00 , and I would like to take over ownership of the ImSwitch GitHub organization and project.
We’re currently working on ImSwitch2 and would like to bring the two projects closer together, making the relationship between them clear, building a bridge and helping users and contributors find their way between them.
We’d be happy to coordinate the handover with you. Thanks again for your contributions and for helping make this transition possible!

beniroquai, 2026-10-13:
sounds great! We have actually diverted a bit from the original fork and have removed its link. Our (old) fork still lives in https://github.com/openuc2/imswitch and is constantly fed with new features. But it has lost all its QT dependencies to run headless with a webui on top on a raspberry pi :)
more information can be found here https://docs.openuc2.com/

xcasas, 2026-10-13:
Hi!!
It's great that you will take over the ownership of ImSwitch organization!! Do you need something from my side? I will grant you the owner rights.

/Xavi

jacopoabramo, 2026-10-13:
Hi @LREIN663 @GuillaumeMinet @khoj00 , if you can confirm that @xcasas gave you ownership rights then this issue can be closed (or since you have those rights now you should be able to do it yourself!)